emoncms/dashboard #3: Dashboard stays 'loading'

Jaco69, 2016-05-13:
After upgrade from v8.5.1 XT to 9.5.1 | 2016.04.28 the dashboard graphs don't show but stay in loading state. See picture:
[image]
I use MySQL for feed values.

RDCH106, 2016-05-27:
Did you update the database from the admin area?

Is the dashboard the only thing not working?

Jaco69, 2016-05-27:
Hi,

Yes I updated the database.
I couldn't get any graph to work, myelektric also didn't work. Except
graphs I haven't found any other problem. All menu's work.

Br,
Jaco

RDCH106, 2016-05-27:
Take a look to the log

closed 2016-07-09 by glynhudson

Jaco69, 2016-12-22:
log is empty

Jaco69, 2016-12-23:
Seems only zoom, simple zoom and the apps are affected, Other graphs are OK. I was only using the zoom on dashboard and my elecric app.
Can it be that MYSQL feeds are broken for zoom?