emoncms/dashboard #27: published and public values don't seem to save anymore with me

joyrider3774, 2016-10-01:
Hi,

i'm on latest pull of both this repo as well as emoncms. I noticed when you are in the dashboard list and you click on the icon's of either published or public the icon changes but the data is not saved for some reason. The sql statement failed when the value was false but not when it was true. 

i could check this by clicking on the publish icon of a dashbard that was already published, it showed the not published icon but if i refreshed the page the dashboard was still public.

i checked the sql statement and when it was having a value of false i tried to do set `published`= '' which gives the sql error '' is not a valid integer value.

i was able to fix it in my local version by chaning those 3 lines to this below so that hard returns 1 or 0 without quotes (so sql sees it as an integer) and explicity setting 0 solved it somehow for me

```        if (isset($fields->public)) $array[] = "`public` = ". (((bool)$fields->public) ? "1":"0") ."";
        if (isset($fields->published)) $array[] = "`published` = ".(((bool)$fields->published) ? "1":"0") ."";
        if (isset($fields->showdescription)) $array[] = "`showdescription` = ".(((bool)$fields->showdescription) ? "1":"0").""; 
```
now everything works again for me

tridiumcontrols, 2016-10-01:
Yep, Experiencing same issue here. Same issue is present in 9.7.2

joyrider3774, 2016-10-03:
This was also due to strict mode being enabled on mysql it seems, i reverted my code changes and disabled strict mode in the settings file from mysql and with the original code this is working also now.

@tridiumcontrols check if this is the case with you as well (strict mode enabled on mysql)  check comment from @chaveiro here https://github.com/emoncms/emoncms/issues/561 

i'm goana close the issue as it works with original code without strictmode on mysql enabled

closed 2016-10-03 by joyrider3774