emoncms/dashboard #36: Floating Palette

Paul-Reed, 2016-10-11:
The undo/redo feature was recently added to emoncms dashboard, but the additional menu icons cannot easily be accommodated in the existing menu bar width. This is particularly obvious for mobile device users (see screenshot below where the 'save button cannot be accessed).
This problem was initially highlighted by @mattjgalloway and carefully considered before merging, however, it was thought beneficial to add this feature first and then resolve the issue shortly afterwards by developing a **floating palette** to replace the menu bar - a concept which was initiated by @firefox7518

I've raised this issue so that it doesn't get overlooked (and before someone suggests grouping/un-grouping dashboard elements... :thinking: )

Paul

[image]

mattjgalloway, 2016-10-12:
Thanks for opening the issue @Paul-Reed.

For reference, #25 is tracking this.

firefox7518, 2016-10-15:
I have closed PR25 and used the new master for integrating the palette. Its in PR #36.

Paul also started a community thread here: https://community.openenergymonitor.org/t/floating-palette-for-dashboard/1755/1

firefox7518, 2016-10-17:
Hi guys, I need some help in fixing the JS code which lets the palette be draggable.

I use the following code:
`

$(document).ready(function() {		    
		    var $body = $('body');
		    var $target = null;
		    var isDraggEnabled = false;

		    $body.on("mousedown", "div", function(e) {
		       
		    	$this = $(this);
		       	isDraggEnabled = $this.data("draggable");

		       	if (isDraggEnabled) {
		       		if(e.offsetX==undefined){
						x = e.pageX-$(this).offset().left;
						y = e.pageY-$(this).offset().top;
					}else{
						x = e.offsetX;
						y = e.offsetY;
					};

					$this.addClass('draggable');
		        	$body.addClass('noselect');
		        	$target = $(e.target);
		       	};
   
		    });
		    
		     $body.on("mouseup", function(e) {
		        $target = null;
		        $body.find(".draggable").removeClass('draggable');
		        $body.removeClass('noselect');
		    });
		    
		     $body.on("mousemove", function(e) {
	            if ($target) {
	                $target.offset({
	                    top: e.pageY  - y,
	                    left: e.pageX - x
	                });
	            };     
		     });

		});

It looking on the mouse event and the DIV which holds the trigger "data-draggable="true" "

However you can move also the buttons in the box and also all entries in the dropdown lists. Thats of course not really nice and good. As said I'm not an experienced programmer and need some help fixing this.

Adding an additional div and set there data-draggable="false" does not help. It still allows to move the buttons. 

Any ideas on that?

mattjgalloway, 2016-10-18:
I'll take a look later and see if I can help out.

firefox7518, 2016-10-18:
Thanks Matt!

I also proceeded and studied a lot on sources like Stack Overflow and google. I found a script which looked neat and seems to do what I wanted. The buttons and icons do not move anymore. The palette acts as expected

I've added the following script to dashboard_edit_view.php:

`<script type="application/javascript">
window.onload = addListeners();
var x_pos = 0,
      y_pos = 0;
function addListeners() {
  document.getElementById('toolbox').addEventListener('mousedown', mouseDown, false);
  window.addEventListener('mouseup', mouseUp, false);
}
function mouseUp() {
  window.removeEventListener('mousemove', divMove, true);
}
function mouseDown(e) {
  var div = document.getElementById('toolbox');
  x_pos = e.clientX - div.offsetLeft;
  y_pos = e.clientY - div.offsetTop;
  window.addEventListener('mousemove', divMove, true);
}
function divMove(e) {
  var div = document.getElementById('toolbox');
  div.style.position = 'absolute';
  div.style.top = (e.clientY - y_pos) + 'px';
  div.style.left = (e.clientX - x_pos) + 'px';
}
</script>`

I named the div "toolbox" for testing, can change it later to palette if the code looks ok for you Matt.

mattjgalloway, 2016-10-18:
@firefox7518 nice one!

I created #41 to tidy up by using jQuery instead of plain JS. Just a bit nicer that way.

Good work here! Looks great.

Paul-Reed, 2016-10-22:
Although further 'palette' development is still continuing, the main feature has now been added to the dashboard repo, so I'll close this with a big thanks to @firefox7518 for his effort & perseverance.

Any further issues can of course be posted individually.

Paul

closed 2016-10-22 by Paul-Reed