emoncms/dashboard #39: Floating palette selects widgets

Paul-Reed, 2016-10-18:
I've noticed that if you drag the palette around the dashboard, it selects (highlights) widgets, menus etc in the dashboard - see screenshot.
[image]

firefox7518, 2016-10-18:
I tried to fix it with my new PR. I added a pauseEvent to it. It seems to solve it as I'm not able anymore to replicate this.

firefox7518, 2016-10-20:
fixed with update from Matts Jquery code

Paul-Reed, 2016-10-20:
Still persists here with Matt's jQuery code applied;

[image]

Paul

firefox7518, 2016-10-21:
Does not happen for me on any browser. Which one are you using?

Paul-Reed, 2016-10-22:
fixed by https://github.com/emoncms/dashboard/pull/48

closed 2016-10-22 by Paul-Reed