emoncms/dashboard #42: Floating Palette - support for tablets

Paul-Reed, 2016-10-18:
The palette can be moved around using a mouse OK, but does not respond to being dragged using a touch device such as a tablet or other mobile device.
The palette could be obstructing a dashboard element, and users would be unable to move it around the workspace.

Paul

firefox7518, 2016-10-18:
he jquery needs also support for touchevents. At the moment its just mousedown etc.

glynhudson, 2016-11-15:
As I have mentioned before do we *really* want to be editing a dashboard on a tablet??! Obviously, the argument could be made for 'on-site' or 'on-couch!' editing. However IMO I don't think it's worth the dev effort. The dashboard editor is complex with lots of small targets, it would be a terrible experience to try and construct a dashboard on a tablet!  Now viewing dashboard is a different story., I think this is where our dev efforts should go... 

https://community.openenergymonitor.org/t/mobile-tablet-dashboard-module/1854
https://community.openenergymonitor.org/t/mobile-app-emoncms-mobile-dash-mockup/2018

pb66, 2016-11-15:
I do not want to get into a debate on this again, but there is a huge range of touch screen devices out there, I purchased a grands worth of 15.6" laptop for using on-site for emoncms and other OEM related dev and promotion, and that has a touch screen. 

Agreed mobile phones and <10" tablets may not be ideal, but with touch screen laptops and desktops becoming more common place it would be foolish to outright exclude it.

My experience with my touchscreen laptop has led me to consider buying a replacement 24" monitor for my desktop with touch, but the anti-touchscreen comments being made here and on the forum are the most prominent "cons" when weighing up the value in that.

@mattjgalloway has already shown interest in making emoncms more touchscreen freindly, prior to the pallette's introduction, but I would imagine he would be less than keen to invest his time in something that's being classed as a waste of time.

@chaveiro recently suggested a "docking" option #58 which would also help with the initial point raised by @Paul-Reed as it wouldn't be obstructing the page. FWIW the pre-pallette toolbar was far more useable on normal laptop sized touchscreens, so I am effectively worse off with the pallette despite thinking it is an excellent idea.

At a later point we could look at adding better touchscreen support, perhaps adding settings for "button size" and "grab handle size" as well as docked or floating pallete.

There are many levels of compromise inbetween "fully implemented right now" and "never", I say leave this issue open for consideration in the (hopefully not so distant) future and allow @firefox7518 to continue.

Paul-Reed, 2016-11-16:
I originally posted this before much of the debate took place about emoncms
& dashboards, but none the less I labelled it as a enhancement - and maybe
something to consider for the future, especially as touch screen devices
become more usable and more common (which no doubt they will).
I agree with @pb66 that this should perhaps be left open, not just as a
reminder, but to show that we've actually debated and considered this
option.

Paul

glynhudson, 2016-11-16:
I'm happy to leave this open. Don't get me wrong I think mobile support is
a good idea. I just think we should prioritise the *viewing* of dashboards
on mobile devices over the ability to create dashboards on mobile  devices.
I also use a touch screen laptop and love using touch, but when it comes to
photo / dashboard editing I always reverted to the mouse / touch pad for
increased precision.