emoncms/dashboard #43: Floating Palette - move palette offscreen

Paul-Reed, 2016-10-18:
_I thought that this had been fixed earlier, but I may be wrong?_

The palette can be dragged off of the dashboard and cannot be recovered without exiting the edit mode by selecting a different menu item, losing any changes that had been made.

Not a major issue! but nice to avoid.

[image]

glynhudson, 2016-11-15:
Haha yes, I've just managed to successfully 'loose' the pallet off the screen which means any changes made cannot be saved. This would be suboptimal if a lot of changes to a dashboard has just been made. If this could be fixed it would be great @firefox7518 ? 

 I love the new buttons btw. They have 'grown' on me 👍

firefox7518, 2016-11-16:
Tried to fix that with PR #60
I think its ok to stop it from moving outside on top and left positions but it should freely float to the right and down as it creates a normal scrollbar and is always accessible.

Let me know if this is ok. this screen position grabbing things are horrible :-) At least for me as a novice.

glynhudson, 2016-11-16:
A good start! This has fixed the constraints in the upwards Y-direction, but it's still possible to move the palette off the screen in the bottom Y-direction on in the right x-direction. Do you see same effect? See my screen cast: https://youtu.be/gwbXd1N-0Ic

Paul-Reed, 2016-11-16:
Can't you use the X & Y scroll bars to retrieve the palette?
I'm not in a position to test, but looking at your video, the scroll bars
appear to be populating.

firefox7518, 2016-11-16:
Glyn, yes, as stated in the PR I just limited the upper and left area because when moving it down and right the scrollbar shows up and has the normal behavior. I would find it a little odd if you have to first place a widget on the dashboard and then move it down or right before you are able to move the palette more down or to the right so I left this open and free.

glynhudson, 2016-11-16:
Ah yes, sorry my bad. Yup, it does what it says on the tin. This is a big improvement since it's now not possible to totally lose it and not be able to save changes. I will merge 👍

closed 2016-11-16 by glynhudson

firefox7518, 2016-11-16:
If you guys think it should not move to the right and down below the dashboard area I of course can bound it too :-)

commit 73bf77cd (2016-11-16): Fixing moving Palette outside of upper and left DIV limits (#60) fixes #43

Added fixed values for stopping the palette from moving of screen on the left and top.
Moving it to the right and down still works and will create a scrollbar. The palette is therefore always reachable.