emoncms/dashboard #50: Dashboard editor blank and unresponsive

CNek, 2016-10-22:
Hello there,
So with EmonCMS 9.7.4 + Fresh clone of dashboard module, on a fresh setup.
I have some inputs feeds kicking in.
I create a new dashboard, then i cannot create new components on the canvas.

I get :

`SyntaxError: missing ) after argument list -  vis_render.js:45:350`

and

```
TypeError: fn is not a function - render.js:44:22
	render_widgets_init *******/render.js:44:22
	<anonyme> https://*****.com/dashboard/edit:249:5
```
Maybe I will try with an older release as it worked for me before on a different installation.

Paul-Reed, 2016-10-22:
Cannot replicate issue using emoncms 9.7.5

Paul

Paul-Reed, 2016-11-05:
Do you still have this issue, what have you tried to resolve it?

glynhudson, 2016-11-15:
I cannot replicate either

closed 2016-11-15 by glynhudson