emoncms/dashboard #58: Floating Palette - should be optional

chaveiro, 2016-11-12:
The floating pallet should have the option to snap to top and have a simoiar behaviour as before.
If space is a constrain, then sap to the side can be an option also. 
By sapping i mean not to be floating at the top/sides but instead be a lateral menu that offsets the dashboard editor window.

Paul-Reed, 2016-11-12:
So users would then have the option to either use the floating panel, or dock it as a menu bar?

chaveiro, 2016-11-12:
Affirmative!

glynhudson, 2016-11-15:
I don't see the problem with the floating palette not being optional. Please post a screenshot illustrating how this is an issue. In my opinion having the requirement to 'snap' the palette back to the top bar will stop any future developments since new buttons won't fit on the top bar which is the reason for the floating pallet development in the first place.  Let along the complexity required to handle the docking and un-docking. I think the development effort required for this would be better spend elsewhere. The floating pallet design was extensively discussed by @firefox7518  : 

https://github.com/emoncms/dashboard/pull/25
https://github.com/emoncms/dashboard/pull/55
and many others...

closed 2016-11-15 by glynhudson

chaveiro, 2016-11-16:
This is a feature request, an enhancement.
Everybody understood what i mean.