emoncms/dashboard #63: Rounding issues for numbers <= 10 in widgets

Paul-Reed, 2016-11-26:
**Test Setup**
Emoncms low-write 9.7.8 | 2016.11.10 with current 'master' version of dashboard installed.
An input fed via browser API calls, which in turn populates a feed - with no other processes defined.

**Results**
Errors occur when the feed value is equal to or less than 10, for example feeding 1, 9, 10 & 11, results in 1.01, 8.99, 9.99 & 11.0 being displayed in the bar & dial widgets, whilst the 'feedvalue' widget (blue text) displays the same value correctly.

[image]
[image]
[image]
[image]

Paul-Reed, 2016-11-27:
@chaveiro tried the fix that you suggested in commit https://github.com/emoncms/dashboard/pull/64 which was changes as follows;
[if ((val * 1).toFixed(2) != (associd[feedid]['value'] * 1).toFixed(2) || redraw == 1)](https://github.com/emoncms/dashboard/blob/master/widget/bar/bar_render.js#L77) to Fixed(3) with partial success...
Although the bar widget now displays the correct values - 9.00, 10.00 & 11.0, the actual values [presented to the 'if' statements](https://github.com/emoncms/dashboard/blob/master/widget/bar/bar_render.js#L243-L254) (to 5 dp) are 9.00059,  9.99922  10.99922.
This then still results in errors, because for example; [else if (raw_value>=10)](https://github.com/emoncms/dashboard/blob/feature/barwidget-rework/widget/bar/bar_render.js#L264) will fail because 'val' is 9.99922 and not 10.
Is this something needing fixing - or something we should just accept ....

chaveiro, 2016-12-08:
Hi Paul, i try to mitigate all this precision issues with https://github.com/emoncms/dashboard/pull/67

Paul-Reed, 2016-12-08:
Thanks for looking at this Nuno. 👍 
Can you recheck the precision rules please, any values greater than 9.99 are displayed as precision instead of to 1 decimal place
For example;
0.99 = 0.99
9.99 = 9.99
11.99 = 11
111.99 = 111

Paul

chaveiro, 2016-12-09:
Hi Paul, i'm not sure what is the problem.
Are you sure of this ?
11.99 = 11  <- i get 12
111.99 = 111  <- i get 112

Paul-Reed, 2016-12-09:
Yes I do now 👍 
I think the problem was due to Cloudflare caching my site, despite putting Cloudflare into development mode & purging the cache last night. This morning I've again purged the server cache and everything now working fine.
I'm going to set up a separate test installation without using Cloudflare for testing purposes as it's the second time I've had this happen.
Thanks again Nuno.

closed 2016-12-09 by Paul-Reed