emoncms/dashboard #68: Dash Background Dont Save

Paulf007, 2016-12-09:
After you have changed the background color of the dash and then save it will show on the new color as long as you are in "edit" mode , but the moment you go to "preview" mode it changes back to default.

Paul-Reed, 2016-12-09:
Cannot replicate here with self-build emoncms low-write 9.7.8 | 2016.11.10 & current dashboard master.
Can you recheck please and provide some information about your system, including versions.

Paul

Paulf007, 2016-12-09:
Hi Paul 
Updated the database through admin.
It is a Windows installed server , server info below:
[image]
Also when I chnage the default color in the "dashboard_schema.php" it does not update on the dash it self. You have to clone the dash then it will update to the new dash. Also noticed that when you made the dash public you cannot change it to private, The setting will show it as private but as soon as you update the page it will revert back to the public setting. 

Hope this gives you enough info. 
Paul

Paul-Reed, 2016-12-09:
It sound as though you've made some local changes to the dashboard module?? If so, can you install a 'stock' version of the dashboard module and see if your problem still exists.
Also, is this a new installation, and if not, what led up to this issue - were there any changes made?

Paulf007, 2016-12-09:
The only thing that I changed was the color value in the dashboard_schema.php , that and a few extra widgets. Could they have an impact?

Paulf007, 2016-12-09:
Ok tested with the stock standard dash module with the same effect , the dash changes to the selected background but the moment you move out of edit mode it reverts back to default. Where is the values for the dash config stored? Maybe I could look what happens there?

Paul-Reed, 2016-12-09:
It's stored in MYSQL 
If you look at the respective  'dashboard' table, you will see column called 'backgroundcolor' which should contain the color code in hex.

Paulf007, 2016-12-10:
Cool I have found it but is that just the default color? Where is the color for every individual dash saved?

Paul-Reed, 2016-12-10:
As per my post above, each dashboard saved background color is saved in it's respective table in the column 'backgroundcolor'.

Paul

Paulf007, 2016-12-11:
Hi Paul

Now my limited knowledge is catching up with me but I would then venture as to say that the updated background value is not passed to the dash file in the Sql. 

I also noticed that once a new dash is created you cannot alter the any of the information for that dash created. The only way to change it is to clone the dash and then rename and change the attributes. Could it be that the "dash" table does not accept alterations once a new line is created with the values added?

Paul-Reed, 2016-12-11:
As per my post above; 'is this a new installation?'
ie, has this ever worked?

Paulf007, 2016-12-11:
Hi Paul 
Sorry I did not make myself clear. 
Yes it is a new install , I never played with the background before and only noticed the problem now. So it would not be able to say if it worked before.  Is there anything else that I could check to assist? All other features works dash design , widgets , feeds inputs... Only problem seems to be the dash table with in the SQL. Here is a screenshot of the table I don't know it that could help. 
[image]

Paulf007, 2016-12-11:
Ok I did some more testing on the Dashboard's 

1) If you create a new blank dash and there is no data in the contents field you are able the change the "public , published values to either 1 or 0 
2) Once there is info in the "content" filed the above values do not change and you need to delete the dash .
3) A New dash background color does not change in the sql even if you save it and it shows on the dash editor. The value does not change in the sql. If you change the value directly in the sql it updates to the dash but not the other way around.  
4) All other features with in the dash do change with out any problems.

Here is some screenshots of the SQL and the dash control:
Dash before edit
[image]
SQL before edit 
[image]
Dash After Edit 
[image]
SQL after Edit
[image]
 If you refresh the page the menu reverts back to origina.

Best regards
Paul

Paul-Reed, 2016-12-11:
All that I can think of is either the MYSQL database privileges or the MYSQL user account have not been set up correctly [as per the guide](https://github.com/emoncms/emoncms/blob/master/docs/WindowsInstall.md).
I don't use a Windows environment, but maybe [post in the forum](https://community.openenergymonitor.org/) linking to this thread, and perhaps a Windows user can help you further.

Paul

Paulf007, 2016-12-12:
Hi Paul 

I will check the privileges in the SQL and give feedback. Also will post on the forum. 
Thank you for taking the time to assist!
Paul

Paul-Reed, 2017-01-02:
Paul, is there a update on this issue please.

Paul

Paulf007, 2017-01-04:
Morning Paul

To be honest with holiday season I have not spend much time on Emon. I will
have a look at it now in the new year to see if I can figure something out.

Best regards

Paul

rjsc, 2017-02-02:
I have performed a new installation in Linux (Ubuntu Server 16.10) and am experiencing the same issue.
Besides not saving the background, any other changes done in the "Dashboard Configuration" window such as Alias, Description or setting as Main are not saved.
However, changing the Alias, or setting as Main in "Setup -> Dashboards" (emoncms/dashboard/list) is saved.

rjsc, 2017-05-10:
I have the following in my apache log file when I try to change the background:
`PHP Notice:  Undefined index: backgroundcolor in /var/www/html/emoncms/Modules/dashboard/Views/dashboard_edit_view.php`
I simply pulled from the repository and didn't change any file besides settings.php

Paul-Reed, 2017-05-10:
Have you updated your MYSQL Database?
Setup > Administration > Update database

Paul

rjsc, 2017-05-11:
@Paul-Reed
My bad. Updated emoncms but not the database.
However the original issue is still present: background is not saved, as any of the settings in "Dashboard Configuration" (grid size, description, main, publiseh, etc). The dashboard content is saved.

rjsc, 2017-05-11:
What bugs me is that changing any configuration in this screen doesn't get saved
[image]
However, any change to the same dashboard in the following screen is saved
[image]

pb66, 2017-05-11:
@rjsc Can you try changing the background colour again, but when you "save" and exit the pop-up "dashboard Configuration" page, can you try clicking the green "not modified" button.

I have seen some pretty erratic behaviour when saving background colours too, is this behaviour consistent every time for you?

rjsc, 2017-05-11:
@pb66 Clicking the green "not modified" button results in the error described in [https://github.com/emoncms/dashboard/issues/31](url)
If I change something in the dashboard itself to get the "changed press to save" button and save it with the background color changed it still gets back to the default blue color.
The behavior is consistent. Saving anything from the "Dashboard Configuration" fails every tiime.

firefox7518, 2017-05-24:
Hmm must be a long existing issue. Even my old 8.4 instance which I did not update since then has this issue. What I get in F12 Chrome console is:
[image]

closed 2017-10-05 by glynhudson