emoncms/dashboard #72: Opening Dashboard on Firefox crashes it.

gitmeister, 2017-02-11:
System is Windows 7 running FF 51 w/ multiprocessing enabled.

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	firefox.exe
  Application Version:	51.0.1.6234
  Application Timestamp:	5888f36c
  Fault Module Name:	xul.dll
  Fault Module Version:	51.0.1.6234
  Fault Module Timestamp:	5888f81b
  Exception Code:	c0000005
  Exception Offset:	00000000003a73a8
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	4105
  Additional Information 1:	724c
  Additional Information 2:	724c5891e4fd2e6702ee69025255ea65
  Additional Information 3:	8678
  Additional Information 4:	8678eadf4d008c91ffef26e1f7b83401

Paul-Reed, 2017-02-11:
Have you tried:

1) clearing the browser cache

2) trying a different browser

Paul

gitmeister, 2017-02-11:
Yes cleared cache on FF - still crashes. Works fine on Chrome.

Paul-Reed, 2017-02-11:
OK, if you start Chrome in developer mode - F12, and see if there are any
significant errors reported in the console. If there isn't, I think it's
safe to assume that it's an issue with your Firefox installation, and not
emoncms, and therefore probably better to report your crash issue to
Firefox devs.

I'm using Firefox here and it's loading fine.

gitmeister, 2017-02-11:
In Chrome just get a warning about JQuery and the "user experience".

Disabled Multiprocessing in FF to check if it was related but reacts the same. Agree it's likely a FF thing.

closed 2017-02-11 by gitmeister