emoncms/dashboard #73: Upgrading the dashboard module on emoncms.org

TrystanLea, 2017-02-14:
@mattjgalloway @chaveiro @RDCH106 Do you have any views on upgrading the dashboard module on emoncms.org, is there anything I should look out for? potential issues to be aware of?

I noticed in brief testing that the dropdown menu's overflow out of the tool box and Im not sure if the double arrow buttons do anything? are they for moving layers to the background/foreground?

Looks great otherwise!

RDCH106, 2017-02-14:
The arrow buttons work perfectly and they are very useful. If I remember correctly @mattjgalloway developed the functionality.

I think that there are not potential issues upgrading the dashboard module.

mattjgalloway, 2017-02-21:
Hmm sounds odd about the overflow and the arrow buttons not doing anything. Do you have a screenshot of the overflow @TrystanLea?

TrystanLea, 2017-02-27:
Thats strange it seems to be working now, dashboards updated on emoncms.org. Thanks a lot and it looks great! :)

TrystanLea, 2017-02-27:
I've removed the curl widget on emoncms.org as I think it would be best to avoid user defined curl requests out from emoncms.org, but I cant seem to find the php server side part of this widget?

RDCH106, 2017-03-01:
@TrystanLea You are right! It could be dangerous for an open access in emoncms.org.

You can remove it safely removing the `curl` folder in `widget` folder.

The full path:
`emoncms\Modules\dashboard\widget\curl `

TrystanLea, 2017-03-01:
Thankks @RDCH106, I've removed that folder, but what confused me was that the content of that folder is just client side javascript, curl needs to run on the server if the requests are directed at an external server. Otherwise a simple AJAX request would fail due to same-origin policy?

RDCH106, 2017-03-01:
@TrystanLea You are right again!

It is not CURL, it is AJAX. The CURL name is symbolic name to denote an easy to use wrapper like the original CURL library.

The browser protect the client side when AJAX response comes from a service without correctly CORS (Cross-Origin Resource Sharing) configuration. But if you only search send information because the response it doesn't matter, the AJAX request continues working.

The basic use of CURL widget is to send for example a value to an emoncms input:

- https://github.com/emoncms/dashboard/pull/4
- https://community.openenergymonitor.org/t/new-curl-widget-available-in-dashboards/554

I implemented CURL widget to work with node-red in mind. I implemented remote actuations using the widget. Because I can see the result in the inputs of emoncms in a short period after the actuation signal from the CURL signal, I don't work with the AJAX response right now.

In the near future I am thinking in manage the response of the AJAX request generated by the CURL widget.

closed 2017-09-04 by glynhudson