emoncms/dashboard #81: Hiding the editing cog when not logged in

pb66, 2017-05-16:
See ["Embedding emoncms into wordpress"](https://community.openenergymonitor.org/t/embedding-emoncms-into-wordpress/4015?u=pb66) on the OEM forum.

It would be nice to only see the dash editing "cog" when logged in.

Paul-Reed, 2017-05-16:
Any thoughts Andreas? @firefox7518 

Paul

firefox7518, 2017-05-17:
Will try to find a solution on the weekend but if someone else already knows how to do it please feel free to add it :-)

firefox7518, 2017-05-24:
https://github.com/emoncms/dashboard/pull/84 Created a PR for my hiding solution.
The icon should not be available now when your are logged out. You have to log in to see it.

Paul-Reed, 2017-05-24:
Fixed by https://github.com/emoncms/dashboard/pull/84

closed 2017-05-24 by Paul-Reed