emoncms/dashboard #82: Aliases not working with public dashboards

pb66, 2017-05-21:
See ["Accessing public dashboard is not possible using alias "](https://community.openenergymonitor.org/t/accessing-public-dashboard-is-not-possible-using-alias/4075?u=pb66)

http://emon/emoncms/dashboard/view?id=4 works publicly and when logged in

http://emon/emoncms/dashboard/view/name only works when logged in

the alias also works ok when used with an apikey eg

http://emon/emoncms/dashboard/view/name&apikey=abc123def456

glynhudson, 2017-05-21:
Thanks for reporting. Is this due to a recent change? I.e did alias work previously?  I can confirm that I can re-create this issue. 

Dashboard URLs work differently on emoncms.og, `dashboard/view` is not needed and alias works e.g. https://emoncms.org/beehive/live

chaveiro, 2017-07-24:
"dashboard/view" is not needed also on local instalations, use "username/alias"

Paul-Reed, 2017-10-28:
No update since last comment 24th July, safe to assume that alias working as intended & close?
// @pb66

pb66, 2017-10-29:
Yes this can be closed.
Chaveiro's comment make total sense as without a userid, dash id or apikey there is no way of determining the actual dash being requested unless we were to make dash aliases unique (across the whole server not per account).

My set up is significantly different now so I cannot test to confirm.

PR #220 (2019-06-20): Feature/issue 82 public alias
fix #82 

public dashboards are now visible to non logged in users via It's alias. eg:
`dashboard/view/solar`

the links to the dashboard from the sidebar or the list of dashboards in `dashboard/list` both use aliases when available. The link can be the shared without an api read key if the dashboard is set to `public`
closed 2019-06-20 by glynhudson