emoncms/dashboard #145: Explore the possibility of using feedname+tag in place of feedid

pb66, 2018-02-27:
See  the https://community.openenergymonitor.org/t/copy-users-on-private-server-including-dashboard-inputs-feeds-graphs-etc/6673?u=pb66 forum thread.

Exclusive use of feedids makes replicating dashboards across accounts very difficult. An option to use Feedname+tag would allow similar functionality to the apps modules.

As pointed out by @TrystanLea, the use of feedids allows the changing feed names to not impact dashboard operation. So perhaps each widget or perhaps better, dashboard, could have a use feedid or use feedname(+tag) option switch.

The "+tag" is used in the device module as the same feednames can be used multiple time in any one account, so using a feedname+tag would allow a particular feed of any given name to be used, rather than feednames needing to be unique to the account, they only need to be unique to the tag.

TrystanLea, 2018-03-15:
I've had a go at making this work and have a good solution I think for the dial widget and visualisations [https://github.com/emoncms/dashboard/tree/tagname_option](https://github.com/emoncms/dashboard/tree/tagname_option).

The visualisations require these modifications in emoncms core:
[https://github.com/emoncms/emoncms/compare/dashboard_tagname_support?expand=1](https://github.com/emoncms/emoncms/compare/dashboard_tagname_support?expand=1)

I havent finished going through every widget to add the conversion from tag:name to feedid support yet but this is an example of what is required for the dial widget:

[https://github.com/emoncms/dashboard/compare/tagname_option?expand=1#diff-cb9b63498661098295e6314bfbc1cf25R553](https://github.com/emoncms/dashboard/compare/tagname_option?expand=1#diff-cb9b63498661098295e6314bfbc1cf25R553)

The option to build a dashboard using tag:name or feedid is set in the dashboard configuration. On switching the feedmode from feedid to tag:name you still need to go through each widget in the dashboard and re-save its feed configuration.

TrystanLea, 2018-03-15:
I've been through the rest of the widgets in the dashboard and added the required tag:name to feedid conversion.

TrystanLea, 2018-03-18:
forum post https://community.openenergymonitor.org/t/portable-emoncms-dashboards-looking-for-testers/6870

closed 2018-06-14 by glynhudson