emoncms/dashboard #149: Edit and/or cut'n'paste dashboard content

pb66, 2018-03-14:
For some time now I have been duplicating dashboards between users on my own emoncms server instance by creating a blank dashboard in a new account and copying the dashboard content from another user account by accessing the mySQL tables directly. All I then need to do is change the config of each widget from within emoncms to point at a valid feed (ie one belonging to the new user not the source user).

This option is not available to emoncms.org  users as they have no access to the mysql and it is less than ideal to recommend users edit the mysql tables directly in any case.

My thoughts on this are, why do we not have a content text box on the dashboard config page

[image]

So that users can edit the html directly or copy'n'paste dashboards in or out to share or duplicate?

It only needs to be a simple, but larger version of the "Description" text box that's linked to the ["Content" field of the dashboard](https://github.com/emoncms/dashboard/blob/master/dashboard_schema.php#L6) table, everything else is already in place.

We can already do this via a backdoor method using "view source" on any webpage, locating the dashboard page content and pasting that into a "paragraph" widget on a new dashboard.

See the [Copying feeds and dashboards over to another online account?](https://community.openenergymonitor.org/t/copying-feeds-and-dashboards-over-to-another-online-account/6827/5?u=pb66) forum thread for more info.

TrystanLea, 2018-03-15:
I've implemented this in the last 3 commits to master, on reflection I should have done it in a feature branch as I've done for the tagname feature.

pb66, 2018-03-15:
Thanks @TrystanLea 

Just tried this and it works as intended, make fife much easier for duplicating dashboards across accounts or even servers. It also means we can now share dashboards on the forum, not just a show and tell, but the code it's self can be cut'n'pasted between users, perhaps with a little sanitation of removing feed ids if users are concerned about sharing them.

I notice your "To view content changes reload editor after saving" note in the editor. Is there a reason the  content cannot be refreshed on save? If we change the background colour in the same config editor, the colour change is implemented on save. What causes the different behavior?

pb66, 2018-03-15:
Is this something that might find it's way to emoncms.org fairly soon?

TrystanLea, 2018-03-18:
I've fixed the reload changes issue and introduced an xss filter, all in the tagname branch.
Would you have a chance to test? Il post a post in the forums to ask for more help testing.

TrystanLea, 2018-03-18:
see: https://community.openenergymonitor.org/t/portable-emoncms-dashboards-looking-for-testers/6870

closed 2018-06-14 by glynhudson