emoncms/dashboard #170: number formatting: thousands separator option

TrystanLea, 2018-09-19:
thousands separator option (commas or spaces), it would make big numbers much easier to read
see: https://community.openenergymonitor.org/t/using-feed-data-in-textbox/8322/23

emrysr, 2018-09-20:
We could expand on the [emoncms>locale.phpL93 `set_lang_by_user()`](https://github.com/emoncms/emoncms/blob/master/locale.php#L93) function to also format numbers. 

This approach would only support the languages emoncms supports and would avoid having to complicate the user settings interface to include all the locale number options for each language/country (see the list output below).

If the required number formatting differs to what is set in the locales, the user profile view would require additional options. eg: 
[image]
 *[proposed additional fields to store number formatting options]*

If we go with the `set_locale()` option, we'd then just need to use [`localeconv()`](http://www.php.net/manual/en/function.localeconv.php) and  [`number_format()`](http://php.net/manual/en/function.number-format.php) to take advantage of the work already done in the locale settings to correctly format numbers without any additional input from the user.

---

## tldr;

> check for installed locales with this command:
``` bash
$ locale -a
```
> I added a new locale like this:
``` bash
$ sudo locale-gen ru_RU.UTF-8 && sudo update-locale
```

I got all the number formatting options for the new locale with this function:

``` php
if (false !== setlocale(LC_ALL, 'ru_RU.UTF-8')) {
    $locale_info = localeconv();
    print_r($locale_info);
}
```

[this example](http://php.net/manual/en/function.localeconv.php) would output a list (if you had the locale installed):

``` 
Array
(
    [decimal_point] => ,
    [thousands_sep] =>  
    [int_curr_symbol] => RUB 
    [currency_symbol] => ₽
    [mon_decimal_point] => ,
    [mon_thousands_sep] =>  
    [positive_sign] => 
    [negative_sign] => -
    [int_frac_digits] => 2
    [frac_digits] => 2
    [p_cs_precedes] => 0
    [p_sep_by_space] => 1
    [n_cs_precedes] => 0
    [n_sep_by_space] => 1
    [p_sign_posn] => 1
    [n_sign_posn] => 1
    [grouping] => Array
        (
            [0] => 3
            [1] => 3
        )

    [mon_grouping] => Array
        (
            [0] => 3
            [1] => 3
        )

)
```

PHP's [`number_format()`](http://php.net/manual/en/function.number-format.php) function already supports different characters for decimal and thousand separators like this:
``` php
$locales = explode(',','da_DK,nl_BE,nl_NL,en_GB,et_EE,fr_FR,de_DE,it_IT,es_ES,cy_GB,en_US,ru_RU,zh_CN');
$number = 7987104012.1203;
foreach($locales as $l) {
  setlocale(LC_ALL, $l.'.UTF-8');
  $locale = localeconv();
  echo "$l\n";
  echo implode("\n",array(
    "number"=>number_format ( $number , $locale['frac_digits'], $locale['decimal_point'], $locale['thousands_sep'] ),
    "currency"=>money_format('%.2n', $number),
    "date"=>strftime("%x")
  ));
  echo "\n\n";
}
```
number, currency and date formatting for each emoncms supported locale (+ en_US, ru_RU & zh_CN )
```
da_DK
7.987.104.012,12
kr. 7.987.104.012,12
20-09-2018

nl_BE
7.987.104.012,12
€ 7 987 104 012,12
20-09-18

nl_NL
7987104012,12
€ 7 987 104 012,12
20-09-18

en_GB
7,987,104,012.12
£7,987,104,012.12
20/09/18

et_EE
7 987 104 012,12
7 987 104 012,12 €
20.09.2018

fr_FR
7 987 104 012,12
7 987 104 012,12 €
20/09/2018

de_DE
7.987.104.012,12
7.987.104.012,12 €
20.09.2018

it_IT
7.987.104.012,12
€ 7.987.104.012,12
20/09/2018

es_ES
7.987.104.012,12
7.987.104.012,12 €
20/09/18

cy_GB
7,987,104,012.12
£7,987,104,012.12
20.09.18

en_US
7,987,104,012.12
$7,987,104,012.12
09/20/2018

ru_RU
7 987 104 012,12
7 987 104 012,12 ₽
20.09.2018

zh_CN
7,987,104,012.12
￥7,987,104,012.12
2018年09月20日

```

pb66, 2018-09-20:
Can the use of a thousands separator please be optional? 

Even if the separator type isn't user defined, can the use of the separator be toggled on/off in the feed widgets, perhaps defaulting to off for backward compatibility, ie avoid everyone's dashes changing on update without permission.

closed 2022-08-13 by TrystanLea