emoncms/dashboard #171: href addresses getting altered in emoncms!

pb66, 2018-10-04:
I have many dashboards that contain images, they range from a simple single image alone on a page to much more complex pages involving multiple images.

Generally these are set up using a simple paragraph widget with some html like this (but they do differ)
```
<div style="text-align: center;">
<a href="/linked_address"><img src="/images/displayed_image.png" style="max-width: 100%; height: auto;" alt="FYI, image alt text is required">
</a></div>
```
this has worked well for a substantial time, recently I updated the sever that was running 9.8.18 and the dashboard module version of that time, to the very latest versions as of now.

I have just tried to change an image and keep getting this error

[image]

all I have done to get that error is change the image url by 1 character (eg "displayed_image.png" to "displayed_image1.png") both images are present and correct, I have even tried copying "displayed_image.png" to "displayed_image1.png" so it is even the same file it is trying to load/locate.

I have since searched every where I could think of and eventually found that all the href entries in the mysql dashboard table has "`&#10;`" appended to the end of the url. eg src="/images/displayed_image.png&#10;".  Armed with this info I have looked at the dashboard configuration content editor (designed for cutting and pasting content between dashes) and found the "`&#10;`" is also shown there, but when viewing the same href url via the paragraph widget configuration box in the dashboard editor, it is filtered and doesn't show the "`&#10;`" even though it is present in the source.

I have now proved this is the cause of the error I see by editing the href url in the content editor to remove the "`&#10;`" it will save and work fine, obviously there doesn't appear to be any change when viewed via the widget editor as the extra "&#10;" doesn't show beforehand.

My original task was to change 1 image on a page of 32 images. I have managed to achieve this by manually editing every one of the 32 image urls (to remove the "`&#10;`" and add the single "1" to the image url in question) as editing any one or any number less than ALL of the links resulted in the same error being raised when saving.

I do not know how those extra chars were added (by emoncms) but I do know that I have dozens of accounts and many dozens of these pages that all need the "`&#10;`" removed from the end of many href urls. Is there a safe automated way to do this on a live server? or could what ever the paragraph widget box does to filter the source be embraced to remove these unwanted extra's when saving a dash?

It's really annoying that the paragraph widget displays correctly when the source is wrong and yet, it says there is "no change" when you try to re-save without adding further changes, if it would just save what it displays (forced even though no change detected) it might remove the "`&#10;`" itself.

In fact, this check for change and block saving if no changes _detected_ has really hindered debugging this issue. If it was possible to open a dashboard for editing and attempt a save without being told there are no changes, it would have raised the error about "invalid content" and that would have rang alarm bells as the dash content was unchanged, the fact it forces me to make changes and only then triggers an error, suggests there is an error with the changes not the original source.

I have to say, having just had to re-edit all the "`&#10;`" strings in this post as they display as "&#10;" without adding the backticks like so

```
"`&#10;`"
```

I do not cherish the idea of trying to do a mysql query to edit by regex!!!

emrysr, 2018-10-23:
hi @pb66,
I've attempted to re-create the above issue. The `&#10;` HTML entity is the linefeed character. Browsers *should* ignore the linefeed character at the end of text nodes or attribute values.

I ran these steps to try and re-create the issue:
-  added your example HTML to a paragraph widget
-  added images to test with
-  Edited the HTML in the `dashboard` table (directly in the database) to include the line feed HTML entity.
- When I viewed or edited the dashboard in emoncms the image seems fine, even with the additional linefeed character
- If I re-saved the widget the unicode linefeed character is passed in the form and then added into the database `content` field instead of the HTML entity - this should happen for any unicode character supported by utf-8
- Tested in firefox and chrome

[screenshot of renedred newline character from HTML entity &amp;#&#xfeff;10;]
[image]

The above screenshot shows a newline starting before the end of the &lt;img&gt; tag `src` attribute:
```
<img src="/assets/image1.jpg
" style="max-width: 100%; height: auto;" alt="">
```
My images still show without issue.?  You should be able to just leave them in the content without issue.... 
> Am I missing the point of this issue..? apologies if I am.

[screenshot of my dashboard test in firefox - top image is the one with the linefeed character included in the src attribute]
[image]

@pb66 Are you still having this issue? or could you explain what I can do to re-create the issue and fix it - thanks.

pb66, 2018-10-23:
Thanks @emrysr for taking a look.

You are correct that the pre-existing images do display without issue and that the paragraph widget editor doesn't show the additional chars and saves without complaining, But the dashboard itself will not re-save following changes unless you go through and remove all the extra chars from the pre-existing image links.

If I never make any changes to the dashboards, it should function ok, but if I want to add another widget, some free text or as I was trying to do when I stumbled across this, change an image, that's when the issue arises.

In my original example, I could save the widget no problem, I could not save the dashboard (with the edited widget) unless I edited all the widgets on the page to remove the extra chars. Saving the widget is only a temporary status, it allows the user to exit the widget editor, but the changes do not stick until the dashboard itself is saved.

emrysr, 2020-01-13:
@pb66 
hi, could you try this pull request to see if this dashboard issue is now fixed. would be good to close this issue if possible.

pb66, 2020-01-13:
Hi @emrysr 
I'm afraid I do not currently have a dev install to test on and I do not experiment on or update my live servers due to past bad experiences, currently running 9.8.31 and sticking with the "if it ain't broke don't fix it" approach.
Obviously I've had to move on in the 15mths since I was faced with the issue and now I cannot even recall what I did to get around it at the time. But what I can remember is that although the resulting characteristics were the same as the recently reported issue (https://community.openenergymonitor.org/t/html-code-no-longer-saving-in-a-dashboard/12929?u=pb66) with the `&nbsp;` entries, there was at that time no problem with `&nbsp;` chars that I recall, I was aware that they were often present in the code but my issue was purely with `&#10;` chars only, so that would lead me to suspect that something else may have changed in the interim perhaps?

Whilst I am unable to test currently, looking at your proposed fix it looks (to the untrained eye) that you are briefly swapping out just the `&nbsp;` chars for the duration of the checks and saving with the `&nbsp;` reintroduced afterwards, is that right? If so, I'm not sure how replacing the `&nbsp;` entries will have any effect on `&#10;` (or any other potentially unacceptable html chars) or am I missing something?

PR #247 (2020-03-12): checked for html characters in model
added some js translations
added check for nbsp and added it back after xss filter
using async function to save dashboard content

fix #171 

[screenshot of 2 tests causing errors - now don't cause errors]
[image]
closed 2020-03-12 by glynhudson