emoncms/dashboard #172: fresh install cant rename dashboards or set any flags

dafyddhirst, 2018-10-07:
Ive just performed an install as per 
https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/readme.md
then installed graph and dashboard
updated database under admin
now i can create dashboards but not rename them or set public flag etc as the changes revert back automatically to what they were.
any ideas?

EA2EGA, 2018-10-08:
> Is this forum thread related?
>
>
> https://community.openenergymonitor.org/t/problem-with-the-basic-dashboard-data/8503?u=pb66
>
> Have you tried setting an alias?

I had the same problem on Debian 9, till I installed the php7.0-mbstring
package with apt.
I was going to report it but I forgot to do it......

Hau idatzi du Paul (notifications@github.com) erabiltzaileak (2018 urr. 7,
ig. (16:25)):

dafyddhirst, 2018-10-08:
I can confirm installing php7.0-mbstring and rebooting fixes the issue :) 
Thanks

glynhudson, 2018-10-25:
This issue only seems to effect php7 since on php 6.7 mbstring used to be built into libapache2-mod-php5. 

I've added php7.0-mbstring to the Emoncms dashboard install guide and the Emoncms core dependency. It's better that mbstring is installed from the beginning since it's probably needed by other modules. 

Thanks for reporting :+

commit c1ae9f62 (2018-10-25): Install php7.0-mbstring fix #172
closed 2018-10-25 by glynhudson