emoncms/dashboard #206: sidebar banch: 'all dashboards' selected

glynhudson, 2019-04-06:
- When testing sidebar branch when any dashboard is selected "All dashbaoards" are also selected:

[image]

- The title of the dashboard menu should be renamed from "Dashboard" > "Dashboards" 

- The "main" dashboard (default) should be selected when the user clicks on the dashboard icon

[image]

Thanks to @bornpin for these suggestions:

https://community.openenergymonitor.org/t/beta-new-emoncms-sidebar-navigation-layout-feedback-request/10435/32?u=glyn.hudson

emrysr, 2019-04-30:
all dashboards link now not incorrectly highlighted:
[image]

the "main dashboard" has been relabeled to "default".
The default dashboard is hightlighted in the list and appears on all pages as the first bookmark in the sidebar
[image]

emrysr, 2019-05-02:
fixed now in master branch

closed 2019-05-02 by emrysr