emoncms/dashboard #209: widgets no longer displayed in embed mode

thib66, 2019-05-03:
When using the following URL  : dashboard/view?id=3&embed=1&apikey=<your key>,
the widgets are no longer displayed.
Looking at the Firefox console this error message is displayed : 
[image]
[image]
In render.js it seems that path is not initially defined.
I do not meet issues when the dansbord is displayed within emoncms interface (with side bar).
Let me know if you need further information.
Regards,

thib66, 2019-05-03:
<details><summary>Server Information</summary>

| | | |
| --- | --- | --- |
|Services||
|Emoncms|Version|10.0.0-beta
||Modules|Administration : Dashboard v2.0.0 : EventProcesses : Feed : Graph v1.2.3 : Input : CoreProcess : Schedule : Time : User : Visualisation
||Git|URL: origin  :  Branch:   :  Describe: 
|Server|OS|Linux 4.14.66-ovh-vps-grsec-zfs-classid
||Host|webm303.start.ha.ovh.net : webm303.start.ha.ovh.net : (178.33.103.27)
||Date|2019-05-03 18:11:44 CEST
||Uptime| 16:11:44 up  2:58,  0 users,  load average: 1,80, 2,09, 9,07
|HTTP|Server|Apache HTTP/1.1 CGI/1.1 443
|MySQL|Version|5.5.30
||Host|saillago8796.mysql.db (10.0.108.199)
||Date|2019-05-03 18:11:44 (UTC 02:00‌​)
||Stats|Uptime: 176572 Threads: 15  Questions: 145353019  Slow queries: 1679
|Memory|RAM|Used: 19,65% Total: 15,55 GB Used: 3,06 GB Free: 12,49 GB
||Swap|Used: 70,05% Total: 513 MB Used: 359,37 MB Free: 153,63 MB
|PHP|Version|7.3.2 (Zend Version 3.3.2)
||Modules|bcmath v7.3.2 : bz2 v7.3.2 : calendar v7.3.2 : cgi-fcgi : Core v7.3.2 : ctype v7.3.2 : curl v7.3.2 : date v7.3.2 : dba v7.3.2 : dom v20031129 : exif v7.3.2 : fileinfo v7.3.2 : filter v7.3.2 : ftp v7.3.2 : gd v7.3.2 : gettext v7.3.2 : gmp v7.3.2 : hash v7.3.2 : iconv v7.3.2 : imagick v3.4.3RC1 : imap v7.3.2 : intl v7.3.2 : ionCube Loader : json v1.7.0 : libxml v7.3.2 : mbstring v7.3.2 : memcached v3.1.0-dev : mongodb v1.5.0-dev : mysqli v7.3.2 : mysqlnd vmysqlnd 5.0.12-dev - 20150407 - $Id: 7cc7cc96e675f6d72e5cf0f267f48e167c2abb23 $ : openssl v7.3.2 : pcre v7.3.2 : PDO v7.3.2 : pdo_mysql v7.3.2 : pdo_pgsql v7.3.2 : pdo_sqlite v7.3.2 : pgsql v7.3.2 : Phar v7.3.2 : posix v7.3.2 : pspell v7.3.2 : redis v3.1.2 : Reflection v7.3.2 : session v7.3.2 : SimpleXML v7.3.2 : soap v7.3.2 : sockets v7.3.2 : sodium v7.3.2 : SPL v7.3.2 : sqlite3 v7.3.2 : standard v7.3.2 : sysvmsg v7.3.2 : sysvsem v7.3.2 : sysvshm v7.3.2 : tokenizer v7.3.2 : wddx v7.3.2 : xml v7.3.2 : xmlreader v7.3.2 : xmlrpc v7.3.2 : xmlwriter v7.3.2 : xsl v7.3.2 : Zend OPcache v7.3.2 : zip v1.15.4 : zlib v7.3.2
</details>
<details><summary>Client Information</summary>

| | | |
| --- | --- | --- |
|HTTP|Browser|Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:66.0) Gecko/20100101 Firefox/66.0
|Screen|Resolution|1280 x 1024
|Window|Size|1263 x 910
</details>

emrysr, 2019-05-09:
Recent changes (2 weeks ago) mean that the `path` variable mentioned in the above error has been moved to the `theme.php`.  I'd missed the `embed.php` that handles the above case..

emrysr, 2019-05-09:
[image]
fixed in emoncms repo: https://github.com/emoncms/emoncms/pull/1261

thib66, 2019-05-09:
Hello, 
I've just applied the modified embed.php and I confirm the widgets are now correctly displayed.
Thanks  for the fix,
Aymeric

closed 2019-05-09 by glynhudson