emoncms/dashboard #210: Undefined function load_language_files in dashboard_menu.php

thekazak, 2019-05-11:
Hi,
I updated emoncms to latest 9.9.9 from master “74f8acaf Merge pull request #1242 from emoncms/master”
together with dashboard module “58d13b9 Merge pull request #208 from thib66/dashboard_localization-03-05-2019”

Now when try to open main emoncms dashboard page it shows error:
 Fatal error: Uncaught Error: Call to undefined function load_language_files() in /var/www/emoncms/Modules/dashboard/dashboard_menu.php:5 Stack trace: #0 /var/www/emoncms/core.php(172): require() #1 /var/www/emoncms/index.php(301): load_menu() #2 {main} thrown in /var/www/emoncms/Modules/dashboard/dashboard_menu.php on line 5

BR
Dimitar

thib66, 2019-05-11:
Hello,
please upgrade emoncms to 10.0.0-beta and let us know if the issue is still there.
when you say that you have tried to open main emoncms dashboard are you using this following URL  : 
/emoncms/dashboard/list ?
Thanks a lot for your answer,
Aymeric

thekazak, 2019-05-12:
Hello,

Yes, /dashboard/list, as well as /dashboard/view...
It looks like the problem is more general, as it has same error for all actually - /app/view, /user/view ... 
I will try with 10.0.0-beta

BR
Dimitar

thekazak, 2019-05-12:
Hi,

Just to confirm, going to master branch 10.0.0-beta fixed the problem

BR
Dimitar

thib66, 2019-05-12:
hello,
thanks for your feedback.
I think that emoncms team can close the issue.
regards,
Aymeric

glynhudson, 2019-05-12:
Thanks, V10.0.0 will be relased to Stable very soon. Please report any issues

closed 2019-05-12 by glynhudson