emoncms/dashboard #242: Refresh onClick? kwhperiod_draw() called too often.

danbates2, 2019-12-20:
There's a dashboard-wide issue I believe.
While editing widgets in particular this is an issue.
When clicking on a widget, kwhperiod_draw() is called twice. When clicking the spanner kwhperiod_draw() is called again. kwhperiod_draw() continues to be called while the spanner config box is open. kwhperiod_draw() is called repetatively while moving and resizing widgets.

Below is the end-section of a widget I've been working on. I've disabled the fastupdate, meaning the issue could be an onClick elsewhere. Perhaps there's a master rendering script which is making the calls.

I think another reason this is problematic, is if the __draw() contains feed.get_value() calls, then each refresh is resulting in at least one call. In the case of kwhperiod_draw(), it's two calls.
A single 'click-drag to move' operation to edit a widget position actions very many feed.get_value() calls.

```
function kwhperiod_init () {
  $('.kwhperiod').html('')
  kwhperiod_draw()
}
function kwhperiod_slowupdate () {
  kwhperiod_draw()
}
function kwhperiod_fastupdate () {
  //kwhperiod_draw();
}
```

Screen recording in Zip.
[dashboard refresh issue2.mp4.zip](https://github.com/emoncms/dashboard/files/3988540/dashboard.refresh.issue2.mp4.zip)

danbates2, 2019-12-20:
I've made a rough fix here, but still needs a look at I think as the timer I've used to stop refreshing too often results in the values jittering around when first loading.. almost a fix.

chaveiro, 2024-04-11:
Duplicate and Fix here https://github.com/emoncms/dashboard/issues/285

chaveiro, 2024-04-13:
Please close it's implemented.

closed 2024-04-23 by TrystanLea