emoncms/dashboard #254: battery font color

Ultimate-Virus, 2020-03-12:
when battery level changes, it also change the background of the texts. So, if we set a dark color for the text it become unreadable when battery is full, and if we set a lighter color its become unreadable during low battery. it can be seen in the below pictures.

dynamic color change is not wanted, may be white border for the text can solve this issue.
[image]
[image]

glynhudson, 2020-03-12:
I agree, I've also noticed this issue. A possible easier fix could just be to show the % value outside at the bottom or top of the battery.

Ultimate-Virus, 2020-03-14:
yes that could be a solutions. but I have another pretty easy solution.
what if we reverse the gradient. like this. 
[image]

glynhudson, 2020-03-15:
Nice, that's good idea. I would be happy to accept a pull request for this.

thib66, 2020-03-30:
Hello,
would you like to reverse systematically the gradient for all the widgets already created (and for the new ones as well), or would you like to have an option in order to choose if the gradient shoud be reversed? 
Thanks a lot for your answer.

closed 2022-08-13 by TrystanLea