emoncms/dashboard #255: Stacked Solar & Stacked graphs are fixed at 5yr view

glynhudson, 2020-03-12:
The stacked Solar & Stacked graphs are fixed at 5yr view, there is not way to adjust this. This makes these views almost useless, unless you have 5yrs of data! This is a shame since these are useful views if a time frame setting was added. 

[image]

TrystanLea, 2020-03-16:
Perhaps loading the feed start date via /feed/getmeta.json would be a good approach here, that way the view initial monthly view would cover the period of data available?

TrystanLea, 2020-03-19:
Fixed in latest master: https://github.com/emoncms/emoncms/commit/6cbe646d91eab708a9af1dcb62b2ac6988e48cff

closed 2020-03-19 by TrystanLea