emoncms/dashboard #274: Trying to get in touch regarding a security issue

zidingz, 2021-07-22:
Hi there,

I couldn't find a `SECURITY.md` in your repository and am not sure how to best contact you privately to disclose a security issue.

Can you add a `SECURITY.md` file with an e-mail to your repository, so that our system can send you the vulnerability details? GitHub suggests that a [security policy](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) is the best way to make sure security issues are responsibly disclosed.

Once you've done that, you should receive an e-mail within the next hour with more info.

Thanks! (cc @huntr-helper)

TrystanLea, 2021-07-22:
Hello @zidingz

TrystanLea, 2021-07-22:
Thanks @zidingz are you able to use the security policy in the emoncms core repository? https://github.com/emoncms/emoncms/blob/master/SECURITY.md not sure if that works with the huntr dev automation?

zidingz, 2021-07-23:
Hey @TrystanLea 

Thanks for replying so quickly.

Unfortunately, since the security report was made against emoncms/dashboard, our system won't be able pick up on a security policy in the core repo. Were a CVE to be issued for this, having the correct the repository and permalinks is a must.

To save you some time, I've opened a PR https://github.com/emoncms/dashboard/pull/275 which includes a clone of core's security policy. Happy to answer any questions you may have!

closed 2021-07-23 by glynhudson