emoncms/dashboard #281: move lots of elements or add space

daniu-de, 2024-01-07:
Hi
I got a dashboard with lots of sensor data on the top and the graphs for each sensor below + the same data again.

Now I added some sensors to my system and would like to extend the list on top. At the moment I do not see another way than to move ALL elements below the list to get space for the extra sensor data ... that is lots of work ... Does anybody see another way?

chaveiro, 2024-04-12:
Fix on 2.4.0 here https://github.com/emoncms/dashboard/pull/284/commits/8055c0f51ef6030e3f9c2192de6a933fb189ec6b
Adds support for select box for drag multiple widgets
See demo:
[image]

chaveiro, 2024-04-13:
Please close it's implemented.

closed 2024-04-25 by TrystanLea